Ultimaker/CuraEngine #1866: [5.3.0] Retraction speeds / length not correctly applied for IDEX

TheThomasD, 2023-03-26:
**Application Version**
5.3.0

**Platform**
Windows

**Steps to Reproduce**
IDEX print with different materials (here: support with E0, part with E1) and different retraction settings (taken from the output in the created gcode):
*Extruder 0*
`
[values]
...
retraction_amount = 3
retraction_extra_prime_amount = 0
retraction_speed = 40
...
switch_extruder_retraction_amount = 5
switch_extruder_retraction_speeds = 40
...
`
*Extruder 1*
`
[values]
...
retraction_amount = 10
retraction_speed = 25
...
switch_extruder_retraction_amount = 15
switch_extruder_retraction_speeds = 25
...
`

**Actual Results**
When a tool change to E0 happens _inside_ a layer, the retraction settings are not taken into account. Example: E0 (un)retract use a feedrate of _F1500_ instead of _F2400_. When the layer change happens while E0 is active, in the new layer the (un)retract speed is correctly used as _F2400_. For E1, always the correct feedrate _F1500_ is used.

*Excerpt from gcode:*
```
;LAYER:0
...                                              ; printing with T0
G1 X120.953 Y133.692 E16.11409
M205 X7 Y7
G1 F2400 E13.11409                               ; correct travel retraction speed/length (F2400, -3mm)
G0 F9000 X125.205 Y138.194
M205 X4 Y4
G1 F2400 E16.11409                               ; correct travel unretraction speed/length (F2400, +3mm)
G1 F1800 X125.815 Y137.75 E16.13919
...
G1 X154.97 Y139.802 E31.27375
M104 S200
G1 F2400 E26.27375                               ; correct tool change retraction speed/length (F2400, -5mm)
G1 F300 Z2.2
G92 E0
T1
G92 E0
M105
M109 S210
M104 T0 S150
G1 F1500 E-10                                    ; OK travel retraction speed/length (F1500, -10mm)
M205 X7 Y7
;MESH:Tecnica Skischuhgummi - scan.stl
G0 F9000 X154.97 Y139.802 Z2.2
G1 F300 Z0.2
G0 F9000 X154.903 Y130.651
M205 X4 Y4
;TYPE:SKIRT
G1 F1500 E0                                      ; correct travel unretraction speed/length (F1500, +10mm)
G1 F1800 X155.828 Y130.664 E0.03077
...                                              ; T1 printing, only (un)retractions with F1500 and +-10mm
;LAYER:1
...                                              ; still T1 printing, only (un)retractions with F1500 and +-10mm
G1 X137.192 Y142.821 E62.94561
M204 S1250
M205 X7 Y7
G0 F9000 X137.433 Y141.915
G1 F1500 E47.94561                               ; correct tool change retraction speed/length (F1500, -15mm)
G1 F300 Z2.4
G92 E0
T0
G92 E0
M105
M109 S200
G1 F1500 E-5                                     ; incorrect: Why is this happening? last tool change retraction was -5. Also, speed is wrong (should be F2400)
G1 F300 Z2.4
;MESH:NOMESH
G0 F9000 X133.569 Y145.406 Z2.4
M204 S1000
M205 X5.5 Y5.5
;TYPE:SUPPORT
G1 F300 Z0.4
G1 F1500 E5                                      ; incorrect speed (should be F2400), correct length (is +5 now)
G1 F1970.4 X133.709 Y145.835 E5.01501
...                                              ; T0 printing
G1 X133.569 Y145.406 E6.8495
M204 S1250
M205 X7 Y7
G1 F1500 E-3.1505                                ; incorrect speed (should be F2400) and length (should be 3mm, but is 10mm)
G0 F9000 X134.074 Y144.703
G0 X128.981 Y138.996
G0 X128 Y139.588
M204 S1000
M205 X5.5 Y5.5
G1 F1500 E6.8495
...                                              ; T0 printing, some more incorrect travel (un)retractions (speed/length taken from T1?)
G1 X156 Y141.072 E13.55518
M204 S1250
M205 X7 Y7
G1 F1500 E3.55518                                ; incorrect travel retraction speed/length (is F1500/10mm, should be F2400/3mm)
G0 F300 X156 Y141.072 Z0.6
G0 F9000 X156.674 Y141.997
G0 X155.862 Y142.586
G0 X153.834 Y142.586
G0 X126.857 Y136.735
G0 X125.527 Y136.734
G0 X124.636 Y137.38
G0 X125.068 Y137.973
;TIME_ELAPSED:141.491006
;LAYER:2
G1 F1500 E10.55518                               ; incorrect? Why is +7mm unretracted? Also, wrong speed (should be F2400)
M106 S170
M204 S1500
;TYPE:SUPPORT
G1 F2400 E13.55518                               ; correct speed/length now; luckily, the two incorrect (un)retractions canceled each other out...
G1 X125.662 Y137.541 E13.57961
...                                              ; T0 continues
G1 X125.068 Y137.973 E15.39214
G1 F2400 E12.39214                               ; correct travel retract speed/length
G0 F9000 X124.56 Y137.275
G0 X125.383 Y136.678
G0 X126.716 Y136.679
G0 X128.716 Y138.483
G0 X128 Y139.514
G1 F2400 E15.39214                               ; correct travel unretract speed/length
G1 X128 Y142.822 E15.50217
...                                              ; T0 continues with correct (un)retract speed/length
G1 X156 Y141.008 E21.92159
G1 F2400 E16.92159                               ; correct tool switch retraction speed/length
G1 F300 Z2.6
G92 E0
T1
G92 E0
G1 F1500 E5                                      ; incorrect: Why is this done? There is no setting that corresponds to this in T1
G1 F300 Z2.6
;MESH:Tecnica Skischuhgummi - scan.stl
G0 F9000 X153.278 Y150.773 Z2.6
;TYPE:WALL-INNER
G1 F300 Z0.6
G1 F1500 E15                                     ; correct tool change unretraction speed/length (F1500, +15mm)
G1 F3600 X153.454 Y150.529 E15.01001
...                                              ; and this continues over the whole print...
```

**Expected results**
consistent use of the retraction settings supplied to the engine

**Additional Information**
I'll try to attach the full gcode afterwards...

GregValiant, 2023-03-26:
Thanks for the report.
I can reproduce this with a regular dual head printer so I don't think the IDEX is the problem.
Someone from the Cura team will take a look.
I opened your gcode file in Cura.  This preview looks entirely wrong.  There is support growing in the middle of nowhere and there is support growing within the print.  I'm not sure what to make of that skirt.
Is this what you saw in your preview?
[image]

TheThomasD, 2023-09-25:
Any chance of this beeing fixed?

GregValiant, 2023-09-26:
I don't work there so I don't know.  That other report was posted at the end of March.
@MariMakes is there any word on this?

MariMakes, 2023-10-19:
Boop! 🦖 
Sorry for the wait, as mentioned in the linked post, I'll bring it up with the team again. 💪