Ultimaker/CuraEngine #1881: Invalid setting '13' is not a valid 'settings.compiler.version' value.

batot1, 2023-05-10:
Hello all,

I'm trying help distribution Archlinux to push Ulimaker/CuraEngine as package ready to install from distribution packages.
I was trying create CuraEngine according to the instructions:
[]( https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source)
I did all the steps according to the instructions.
But for this moment I was stopped in this place:
```
$ conan install . --build=missing --update
ERROR: Invalid setting '13' is not a valid 'settings.compiler.version' value.
Possible values are ['4.1', '4.4', '4.5', '4.6', '4.7', '4.8', '4.9', '5', '5.1', '5.2', '5.3', '5.4', '5.5', '6', '6.1', '6.2', '6.3', '6.4', '6.5', '7', '7.1', '7.2', '7.3', '7.4', '7.5', '8', '8.1', '8.2', '8.3', '8.4', '9', '9.1', '9.2', '9.3', '9.4', '10', '10.1', '10.2', '10.3', '11', '11.1', '11.2', '11.3', '12', '12.1', '12.2']
Read "http://docs.conan.io/en/latest/faq/troubleshooting.html#error-invalid-setting"

```
I have fresh actual system with all packages (pacman -Syu).
I'm not sure what next doing, maybe somebody can help me?

batot1, 2023-05-10:
Ok I was doing trick resolve this step.
Now compilation all.

But I get many warrning abut maybe 50 example:
/.conan/data/protobuf/3.21.4/_/_/build/cd5b48acbf6628ad1e42a0c12cb321217a0989e7/source_subfolder/src/google/protobuf/repeated_ptr_field.h:1963:5: warning: type attributes ignored after type is already defined [-Wattributes]

Or other example:
CMake Warning (dev) at CMakeLists.txt:2 (project):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done (0.5s)
-- Generating done (0.0s)
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_POLICY_DEFAULT_CMP0091

Is it all normal?

rburema, 2023-05-12:
Yes, the warnings are normal.

Conan has written the build, and you can proceed with the next steps of the instructions, as in, you should be able to call cmake, as described in the 2nd part of step 3, then test if it all is working with step 4 and 5.

batot1, 2023-05-15:
Thanks for answer me @rburema .
"cmake --preset release" - was ok
but:
"cmake --build --preset release" - giving me error
`[121/184] Linking CXX executable tests/ClipperTest
FAILED: tests/ClipperTest
: && /usr/bin/c++ -m64 -O3 -DNDEBUG -m64 tests/CMakeFiles/ClipperTest.dir/main.cpp.o tests/CMakeFiles/ClipperTest.dir/ClipperTest.cpp.o -o tests/ClipperTest  -Wl,-rpath,/home/bartosz/.conan/data/clipper/6.4.2/_/_/package/423bb9bdb88c6c77ef0c088ee84b359c2e60659e/lib:/home/bartosz/.conan/data/arcus/5.2.2/_/_/package/0a575a048007e2b2acbb9ce7fc89813f4189c75c/lib:/home/bartosz/.conan/data/protobuf/3.21.4/_/_/package/cd5b48acbf6628ad1e42a0c12cb321217a0989e7/lib  lib_CuraEngine.a  tests/libtest_helpers.a  lib_CuraEngine.a  /home/bartosz/.conan/data/spdlog/1.10.0/_/_/package/9910850add5893565f188f0dbc3adf0e4db4799e/lib/libspdlog.a  /home/bartosz/.conan/data/fmt/9.0.0/_/_/package/6101eb763716ef74d2c7ee47a67d75a0b37c614d/lib/libfmt.a  -lm  /home/bartosz/.conan/data/clipper/6.4.2/_/_/package/423bb9bdb88c6c77ef0c088ee84b359c2e60659e/lib/libpolyclipping.so  /home/bartosz/.conan/data/arcus/5.2.2/_/_/package/0a575a048007e2b2acbb9ce7fc89813f4189c75c/lib/libArcus.so  /home/bartosz/.conan/data/protobuf/3.21.4/_/_/package/cd5b48acbf6628ad1e42a0c12cb321217a0989e7/lib/libprotoc.so  /home/bartosz/.conan/data/protobuf/3.21.4/_/_/package/cd5b48acbf6628ad1e42a0c12cb321217a0989e7/lib/libprotobuf.so  -lpthread  /home/bartosz/.conan/data/zlib/1.2.12/_/_/package/c11224e931af7eb47df7c689b7ad7afc6b151bd9/lib/libz.a && :
/usr/bin/ld: tests/CMakeFiles/ClipperTest.dir/ClipperTest.cpp.o: in function `cura::DISABLED_ClipperTest_PolylinesTest2_Test::TestBody()':
ClipperTest.cpp:(.text+0x493): undefined reference to `testing::internal::EqFailure(char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
/usr/bin/ld: ClipperTest.cpp:(.text+0x5a4): undefined reference to `testing::Message::Message()'
/usr/bin/ld: ClipperTest.cpp:(.text+0x5d6): undefined reference to `testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)'
/usr/bin/ld: ClipperTest.cpp:(.text+0x5e1): undefined reference to `testing::internal::AssertHelper::operator=(testing::Message const&) const'
/usr/bin/ld: ClipperTest.cpp:(.text+0x5e9): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
/usr/bin/ld: ClipperTest.cpp:(.text+0x6a5): undefined reference to `testing::AssertionSuccess()'
/usr/bin/ld: tests/CMakeFiles/ClipperTest.dir/ClipperTest.cpp.o: in function `cura::DISABLED_ClipperTest_PolylinesTest1_Test::TestBody()':
ClipperTest.cpp:(.text+0x8e3): undefined reference to `testing::internal::EqFailure(char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)'
/usr/bin/ld: ClipperTest.cpp:(.text+0x9f4): undefined reference to `testing::Message::Message()'
/usr/bin/ld: ClipperTest.cpp:(.text+0xa26): undefined reference to `testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)'
/usr/bin/ld: ClipperTest.cpp:(.text+0xa31): undefined reference to `testing::internal::AssertHelper::operator=(testing::Message const&) const'
/usr/bin/ld: ClipperTest.cpp:(.text+0xa39): undefined reference to `testing::internal::AssertHelper::~AssertHelper()'
/usr/bin/ld: ClipperTest.cpp:(.text+0xaf5): undefined reference to `testing::AssertionSuccess()'
/usr/bin/ld: tests/CMakeFiles/ClipperTest.dir/ClipperTest.cpp.o: in function `testing::internal::TestFactoryImpl<cura::DISABLED_ClipperTest_PolylinesTest2_Test>::CreateTest()':
ClipperTest.cpp:(.text._ZN7testing8internal15TestFactoryImplIN4cura40DISABLED_ClipperTest_PolylinesTest2_TestEE10CreateTestEv[_ZN7testing8internal15TestFactoryImplIN4cura40DISABLED_ClipperTest_PolylinesTest2_TestEE10CreateTestEv]+0x17): undefined reference to `testing::Test::Test()'
/usr/bin/ld: tests/CMakeFiles/ClipperTest.dir/ClipperTest.cpp.o: in function `testing::internal::TestFactoryImpl<cura::DISABLED_ClipperTest_PolylinesTest1_Test>::CreateTest()':
ClipperTest.cpp:(.text._ZN7testing8internal15TestFactoryImplIN4cura40DISABLED_ClipperTest_PolylinesTest1_TestEE10CreateTestEv[_ZN7testing8internal15TestFactoryImplIN4cura40DISABLED_ClipperTest_PolylinesTest1_TestEE10CreateTestEv]+0x17): undefined reference to `testing::Test::Test()'
/usr/bin/ld: tests/CMakeFiles/ClipperTest.dir/ClipperTest.cpp.o: in function `_GLOBAL__sub_I__ZN4cura20DISABLED_ClipperTest18intersectPolylinesERKSt6vectorIS1_IN10ClipperLib8IntPointESaIS3_EESaIS5_EES9_':
ClipperTest.cpp:(.text.startup+0x162): undefined reference to `testing::internal::MakeAndRegisterTestInfo(char const*, char const*, char const*, char const*, testing::internal::CodeLocation, void const*, void (*)(), void (*)(), testing::internal::TestFactoryBase*)'
/usr/bin/ld: ClipperTest.cpp:(.text.startup+0x228): undefined reference to `testing::internal::MakeAndRegisterTestInfo(char const*, char const*, char const*, char const*, testing::internal::CodeLocation, void const*, void (*)(), void (*)(), testing::internal::TestFactoryBase*)'
/usr/bin/ld: tests/CMakeFiles/ClipperTest.dir/ClipperTest.cpp.o: in function `cura::DISABLED_ClipperTest_PolylinesTest2_Test::~DISABLED_ClipperTest_PolylinesTest2_Test()':
[... cut ...]
/usr/bin/ld: tests/CMakeFiles/TimeEstimateCalculatorTest.dir/main.cpp.o: in function `main':
main.cpp:(.text.startup+0xe): undefined reference to `testing::InitGoogleTest(int*, char**)'
/usr/bin/ld: main.cpp:(.text.startup+0x13): undefined reference to `testing::UnitTest::GetInstance()'
/usr/bin/ld: main.cpp:(.text.startup+0x1b): undefined reference to `testing::UnitTest::Run()'
collect2: error: ld returned 1 exit status
[130/184] Building CXX object tests/CMakeFiles/GCodeExportTest.dir/GCodeExportTest.cpp.o
ninja: build stopped: subcommand failed.
`

rburema, 2023-05-15:
Did you follow the release or the debug steps? It seems like it's trying to build the unit-tests, but those where only enabled in the Debug example steps, while you're attempting to do a release build.

You might have to re-run with either `... -o curaengine:enable_testing=True` or `... -o curaengine:enable_testing=False` depending.

batot1, 2023-05-15:
No, because debag is next command, past 2 commands cmake.

Now trying:
"conan install . --build=missing --update -s build_type=Debug -o curaengine:enable_testing=True" - running ok.
"cmake --preset debug" - running ok.

"cmake --build --preset release" - still giving me fault, but other fault.
```

/usr/bin/ld: tests/CMakeFiles/GCodeExportTest.dir/GCodeExportTest.cpp.o:(.data.rel.ro._ZTVN7testing13TestWithParamImEE[_ZTVN7testing13TestWithParamImEE]+0x28): undefined reference to `testing::Test::TearDown()'
/usr/bin/ld: tests/CMakeFiles/GCodeExportTest.dir/main.cpp.o: in function `main':
main.cpp:(.text.startup+0xe): undefined reference to `testing::InitGoogleTest(int*, char**)'
/usr/bin/ld: main.cpp:(.text.startup+0x13): undefined reference to `testing::UnitTest::GetInstance()'
/usr/bin/ld: main.cpp:(.text.startup+0x1b): undefined reference to `testing::UnitTest::Run()'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
[10/56] Building CXX object tests/CMakeFiles/ArcusCommunicationTest.dir/arcus/ArcusCommunicationTest.cpp.o
ninja: build stopped: subcommand failed.
```

MajliTech, 2023-07-14:
> No, because debag is next command, past 2 commands cmake.
> 
> Now trying: "conan install . --build=missing --update -s build_type=Debug -o curaengine:enable_testing=True" - running ok. "cmake --preset debug" - running ok.
> 
> "cmake --build --preset debug" - now is ok
> 
> Thanks

Either build release, or debug. Don't build both, since for what? You're probably good to go with Release.

jellespijker, 2023-09-07:
This should be fixed once the following PR https://github.com/Ultimaker/Cura/pull/16607 are merged and you update to conan 1.60.2 which will have version GCC-13 in the settings.yml

MajliTech, 2023-09-07:
> This should be fixed once the following PR > Ultimaker/Cura#16607 <https://github.com/Ultimaker/Cura/pull/16607>>  are merged and you update to conan 1.60.2 which will have version GCC-13 in the settings.yml

Okay.

jellespijker, 2023-09-12:
The PR has been merged. Make sure you use Conan 1.58 or higher, I would just use 1.60.2 (latest v1)

The error you were seeing was because you probably have gcc-13 installed with Conan 1.56, so version 13 was not yet used in the `~/.conan/settings.yml` updating Conan and removing the old settings.yml should be enough.

Alternatively you can add 13 to the GCC version list in the settings.yml manually, or use GCC-12.

Also be aware that the used dependencies changed a lot now, since we're adding gRPC functionality to the engine.

jellespijker, 2023-09-12:
Closing issue because I'm fairly confident that this fixes your issue. If not the case feel free to reopen

closed 2023-09-12 by jellespijker