Ultimaker/CuraEngine #1884: Arcus error

flashstacks06, 2023-05-28:
arcus/5.2.2: Not found in local cache, looking in remotes...
arcus/5.2.2: Trying with 'conan-center'...
ERROR: Failed requirement 'arcus/5.2.2' from 'conanfile.py (curaengine/None)'
ERROR: Unable to find 'arcus/5.2.2' in remotes

cody-the-casual-dev, 2023-06-11:
I was able to get past this by rerunning  the `conan config install..` step, followed by the `conan profile new default...`

Then DONT remove the cura remote as the documentation suggests

jellespijker, 2023-06-11:
good that you fixed it, i will close the issue.

can you point me towards the documentation wherw it still said that you need to remove the cura remote, because I couldn't find it anymore https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source

closed 2023-06-11 by jellespijker

cody-the-casual-dev, 2023-06-11:
I am not the one who opened the issue, so hopefully my suggestion fixed it for OP. 

The instruction is here: https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source#1-configure-conan

It says:

> Community developers would have to remove the Conan cura repository because it requires credentials (conan remote remove cura).

jellespijker, 2023-06-11:
thx I must have missed that, documentation has been updated

zhangjiangzhao123, 2023-08-02:
I encountered the same problem, but I tried to re execute the command 
`conan config install https://github.com/ultimaker/conan-config.git`
`conan profile new default -- detect -- force `
does not solve the problem.