Ultimaker/CuraEngine #1895: [5.x.x] CuraEngine crush when slicing walls

Dogm, 2023-07-05:
**Application Version**
all 5.x.x

**Platform**
Tested on Windows but most likely relevant for all platforms

**Steps to Reproduce**
1) use model.stl - (need rename jpg to stl for correct file format)
[image]

2) move model on platform
x += 1.5005f
y += 1.5005f
z += -0.122840881f

3) set line width = 5.6 mm, layer height 0.2 mm

SlicerProcess will crush with reorting progress of generating WallToolPaths (i have this one in 13 layer (~2.6mm from bottom))

Crush in \src\SkeletalTrapezoidationGraph.cpp

```
void SkeletalTrapezoidationGraph::collapseSmallEdges(coord_t snap_dist) {
...
        //  o-o
        //  | | > collapse sides
        //  o o
        if (should_collapse(quad_start->from, quad_end->to) && should_collapse(quad_start->to, quad_end->from))
        { // Collapse start and end edges and remove whole cell
        ...
            quad_start->twin->twin = quad_end->twin;
            quad_end->twin->twin = quad_start->twin;  // <------ quad_end->twin = null
        ...
        }
...
}
```

Dogm, 2023-07-05:
Any ideas how to fix this bug?

jellespijker, 2023-07-05:
this is the infamous slicing crash that you encounter. It is a complex problem to solve, what often helpa is rotating the model and slice again. 

@MariMakes do we have a Cura issue where we can link this issue?

Dogm, 2023-07-07:
Have same problem sometimes on generate infill walls (Lines pattern)  8(

jellespijker, 2023-07-11:
hi @Dogm I'm closing this issue as a duplicate of Ultimaker/Cura#12019

closed 2023-07-11 by jellespijker