Ultimaker/CuraEngine #1905: ERROR: Invalid setting '13' is not a valid 'settings.compiler.version' value.

MajliTech, 2023-07-14:
**Application Version** Checked out to 2.7.0

**Platform** Fedora
```sh
$ uname -a
Linux fedora 6.3.12-200.fc38.x86_64 #1 SMP PREEMPT_DYNAMIC Thu Jul  6 04:05:18 UTC 2023 x86_64 GNU/Linux
```

**Steps to Reproduce**
1. Follow the official build guide
2. Get stuck at `conan install . --build=missing --update `

**Actual Results**
The build fails with log:
```sh
$  conan install . --build=missing --update                                  
Trying to clone repo: https://github.com/ultimaker/conan-config.git
Repo cloned!
Copying file .gitignore to /home/majlitech/.conan/.
Skip README.md
Processing conan.conf
Copying file global.conf to /home/majlitech/.conan/.
Defining remotes from remotes.txt
Copying file GitHubActionsBuildEnv.py to /home/majlitech/.conan/generators
Copying file GitHubActionsRunEnv.py to /home/majlitech/.conan/generators
Copying file PyCharmRunEnv.py to /home/majlitech/.conan/generators
Copying file VirtualPythonEnv.py to /home/majlitech/.conan/generators
Copying file cura.jinja to /home/majlitech/.conan/profiles
Copying file cura_build.jinja to /home/majlitech/.conan/profiles
ERROR: Invalid setting '13' is not a valid 'settings.compiler.version' value.
Possible values are ['4.1', '4.4', '4.5', '4.6', '4.7', '4.8', '4.9', '5', '5.1', '5.2', '5.3', '5.4', '5.5', '6', '6.1', '6.2', '6.3', '6.4', '6.5', '7', '7.1', '7.2', '7.3', '7.4', '7.5', '8', '8.1', '8.2', '8.3', '8.4', '9', '9.1', '9.2', '9.3', '9.4', '10', '10.1', '10.2', '10.3', '11', '11.1', '11.2', '11.3', '12', '12.1', '12.2']
Read "http://docs.conan.io/en/latest/faq/troubleshooting.html#error-invalid-setting"
```

**Expected results**
The build should go through

**Additional Information**
Trying to build release

jellespijker, 2023-07-15:
There is a mismatch between the version of the compiler on your system and the one specified in conan settings. Due to various reasons we can't update to conan 2.0 yet,  but what you could try to do update the settings.yml file.

Assuming you're compiling with GCC you could try to add `"13"` to `compiler --> gcc --> version` list in the file `~/.conan/settings.yml` and rerun `conan profile new default --force --detect`

MajliTech, 2023-07-15:
The build now goes thru, thanks.

closed 2023-07-15 by MajliTech