Ultimaker/CuraEngine #1937: with libfmt10 build fail

Pro-pra, 2023-08-15:
Hi all, after update libfmt build fail (sorry, builder replace some symbol to artefacts):
```
FAILED: CMakeFiles/_CuraEngine.dir/src/LayerPlanBuffer.cpp.o 
/usr/bin/g++ -DARCUS -DSPDLOG_FMT_EXTERNAL -DVERSION=\"5.3.0\" -I/usr/include/stb -I/usr/include/polyclipping -I/builddir/build/BUILD/CuraEngine-5.3.0/build -I/builddir/build/BUILD/CuraEngine-5.3.0/include -O2 -fomit-frame-pointer -gdwarf-4 -Wstrict-aliasing=2 -pipe -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -fstack-protector-strong --param=ssp-buffer-size=4 -m64 -mtune=generic  -O3 -DNDEBUG -Wall -fopenmp -std=gnu++20 -MD -MT CMakeFiles/_CuraEngine.dir/src/LayerPlanBuffer.cpp.o -MF CMakeFiles/_CuraEngine.dir/src/LayerPlanBuffer.cpp.o.d -o CMakeFiles/_CuraEngine.dir/src/LayerPlanBuffer.cpp.o -c /builddir/build/BUILD/CuraEngine-5.3.0/src/LayerPlanBuffer.cpp
In file included from /usr/include/spdlog/fmt/fmt.h:31,
                 from /usr/include/spdlog/common.h:50,
                 from /usr/include/spdlog/spdlog.h:12,
                 from /builddir/build/BUILD/CuraEngine-5.3.0/src/LayerPlanBuffer.cpp:4:
/usr/include/fmt/core.h: In instantiation of вЂconstexpr decltype (ctx.begin()) fmt::v10::detail::parse_format_specs(ParseContext&) [with T = cura::LayerIndex; ParseContext = compile_parse_context<char>; decltype (ctx.begin()) = const char*]вЂ™:
/usr/include/fmt/core.h:2641:51:   required from here
/builddir/build/BUILD/CuraEngine-5.3.0/src/LayerPlanBuffer.cpp:85:21:   in вЂconstexprвЂ™ expansion of вЂfmt::v10::basic_format_string<char, const cura::LayerIndex&>("Layer {} is empty (or it has empty extruder plans). Temperature control and cross layer travel moves might suffer!")вЂ™
/usr/include/fmt/core.h:2742:40:   in вЂconstexprвЂ™ expansion of вЂfmt::v10::detail::parse_format_string<true, char, format_string_checker<char, cura::LayerIndex> >(((fmt::v10::basic_format_string<char, const cura::LayerIndex&>*)this)->fmt::v10::basic_format_string<char, const cura::LayerIndex&>::str_, fmt::v10::detail::format_string_checker<char, cura::LayerIndex>(fmt::v10::basic_string_view<char>(((const char*)s))))вЂ™
/usr/include/fmt/core.h:2525:36:   in вЂconstexprвЂ™ expansion of вЂfmt::v10::detail::parse_replacement_field<char, format_string_checker<char, cura::LayerIndex>&>(p, end, (* & handler))вЂ™
/usr/include/fmt/core.h:2459:33:   in вЂconstexprвЂ™ expansion of вЂ(& handler)->fmt::v10::detail::format_string_checker<char, cura::LayerIndex>::on_replacement_field((& handler)->fmt::v10::detail::format_string_checker<char, cura::LayerIndex>::on_arg_id(), begin)вЂ™
/usr/include/fmt/core.h:2634:20:   in вЂconstexprвЂ™ expansion of вЂ((fmt::v10::detail::format_string_checker<char, cura::LayerIndex>*)this)->fmt::v10::detail::format_string_checker<char, cura::LayerIndex>::on_format_specs(id, begin, begin)вЂ™
/usr/include/fmt/core.h:2550:45: error: вЂfmt::v10::detail::type_is_unformattable_for<cura::LayerIndex, char> _вЂ™ has incomplete type
 2550 |     type_is_unformattable_for<T, char_type> _;
      |                                             ^
/usr/include/fmt/core.h: In instantiation of вЂconstexpr fmt::v10::detail::value<Context> fmt::v10::detail::make_arg(T&) [with bool PACKED = true; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; T = const cura::LayerIndex; typename std::enable_if<PACKED, int>::type <anonymous> = 0]вЂ™:
/usr/include/fmt/core.h:1810:51:   required from вЂconstexpr fmt::v10::format_arg_store<Context, Args>::format_arg_store(T& ...) [with T = {const cura::LayerIndex}; Context = fmt::v10::basic_format_context<fmt::v10::appender, char>; Args = {cura::LayerIndex}]вЂ™
/usr/include/fmt/core.h:1828:18:   required from вЂconstexpr fmt::v10::format_arg_store<Context, typename std::remove_cv<typename std::remove_reference<_Args>::type>::type ...> fmt::v10::make_format_args(T& ...) [with Context = basic_format_context<appender, char>; T = {const cura::LayerIndex}]вЂ™
/usr/include/spdlog/logger.h:374:75:   required from вЂvoid spdlog::logger::log_(spdlog::source_loc, spdlog::level::level_enum, spdlog::string_view_t, Args&& ...) [with Args = {const cura::LayerIndex&}; spdlog::string_view_t = fmt::v10::basic_string_view<char>]вЂ™
/usr/include/spdlog/logger.h:90:13:   required from вЂvoid spdlog::logger::log(spdlog::source_loc, spdlog::level::level_enum, fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const cura::LayerIndex&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const cura::LayerIndex&>]вЂ™
/usr/include/spdlog/logger.h:96:12:   required from вЂvoid spdlog::logger::log(spdlog::level::level_enum, fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const cura::LayerIndex&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const cura::LayerIndex&>]вЂ™
/usr/include/spdlog/logger.h:164:12:   required from вЂvoid spdlog::logger::warn(fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const cura::LayerIndex&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const cura::LayerIndex&>]вЂ™
/usr/include/spdlog/spdlog.h:176:31:   required from вЂvoid spdlog::warn(fmt::v10::format_string<T ...>, Args&& ...) [with Args = {const cura::LayerIndex&}; fmt::v10::format_string<T ...> = fmt::v10::basic_format_string<char, const cura::LayerIndex&>]вЂ™
/builddir/build/BUILD/CuraEngine-5.3.0/src/LayerPlanBuffer.cpp:85:21:   required from here
/usr/include/fmt/core.h:1577:63: error: вЂfmt::v10::detail::type_is_unformattable_for<const cura::LayerIndex, char> _вЂ™ has incomplete type
 1577 |     type_is_unformattable_for<T, typename Context::char_type> _;
      |                                                               ^
/usr/include/fmt/core.h:1581:7: error: static assertion failed: Cannot format an argument. To make type T formattable provide a formatter<T> specialization: https://fmt.dev/latest/api.html#udt
 1581 |       formattable,
      |       ^~~~~~~~~~~
```

Pro-pra, 2023-08-15:
src/LayerPlanBuffer.cpp need patch for new libfmt10?

jellespijker, 2023-08-15:
we use fmt 9

https://github.com/Ultimaker/CuraEngine/blob/main/conanfile.py#L82

I'm not sure how it came to be that you're using fmt 10.

did you install the dependencies with Conan according to our build instructions?

Pro-pra, 2023-08-16:
from fedora patch:
https://src.fedoraproject.org/rpms/CuraEngine/blob/rawhide/f/CuraEngine-5.3.0-fmt10.patch

jellespijker, 2023-08-16:
thanks for the patch, I don't think that is the solution that we will implement, though, it make more sense to use a Custom formatter capable of handling these LayerIndex's in the same way as that we do the thread_id here https://github.com/Ultimaker/CuraEngine/blob/CURA-10475_engineplugin/include/utils/format/thread_id.h 

Since we still use fmt 9 and not 10 I don't think that we implement this now, but Pull Request are welcome.

And if you install this with `conan install ...` it should use fmt 9 by default

closed 2023-08-17 by Pro-pra