Ultimaker/CuraEngine #1938: How to find Visual studio solution and build it

actualyang, 2023-08-17:
Follow the building wiki, however can't find the sln file. Besides, If I open the folder of curaEngine in VS2022 it popup the cmake error :
	CMake Error at H:\Code\CuraEngine-main\CMakeLists.txt:7 (find_package):
  By not providing "Findstandardprojectsettings.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "standardprojectsettings", but CMake did not find one.

  Could not find a package configuration file provided by
  "standardprojectsettings" with any of the following names:

    standardprojectsettingsConfig.cmake
    standardprojectsettings-config.cmake

  Add the installation prefix of "standardprojectsettings" to
  CMAKE_PREFIX_PATH or set "standardprojectsettings_DIR" to a directory
  containing one of the above files.  If "standardprojectsettings" provides a
  separate development package or SDK, be sure it has been installed.	CuraEngine	H:\Code\CuraEngine-main\CMakeLists.txt	7	

Is there is sln file if I following the build guild. 
If no sln file how to build it.

jellespijker, 2023-08-17:
I'm not a visual studio guy myself but I think the answer might be here: https://learn.microsoft.com/en-us/cpp/build/cmake-projects-in-visual-studio?view=msvc-170

maybe @rburema can help you if your still stuck

rburema, 2023-08-17:
@jellespijker is correct. If you've set up C++ for Visual Studio, the CMake option should be on by default. (As explained in the link.)

There isn't a 'solution' file (`*.sln`) when using a CMake workflow with Visual Studio. When in VS, you should be able to press 'build all' after doing the Conan steps.

Are you using PowerShell or the 'x64 Native Tools Command Prompt for VS' (possibly with PowerShell activated within it)? I've found that the latter always works.

Possibly you need to not skip the first line (`cmake --preset debug`) of the steps to be skipped on Windows either.

actualyang, 2023-08-18:
Thank your two guys. I will try it.

actualyang, 2023-08-19:
Still can't pass if no cmakelist edit.
I fix it before the find_package... with set(CMAKE_PREFIX_PATH "H:\\Code\\CuraEngine-main\\build\\generators").

closed 2023-08-19 by actualyang