Ultimaker/CuraEngine #1966: Can I modify the slicing algorithm based on the Curaengine source code？

biexiangi, 2023-10-08:
As a graduate student, my current research direction is the secondary development of cura software and innovation in its slicing algorithm. At present, according to the source code of Cura and Curaengine, I have connected external Curaengine through Cura to conduct external debugging. However, I still don't understand the algorithm involved in the Curaengine source code, so I want to ask the feasibility of slicing algorithm innovation through two source codes, and whether I can complete my project. As a novice, I have been working hard. I hope someone can give me a simple answer. thank you
**Application Version**

all latest from main branch
**Platform**

Win11
**Qt**

None
**PyQt**

PyQt6
**Display Driver**

none
**Steps to Reproduce**

none
**Actual Results**

none
**Expected results**

none
**Additional Information**

none

jellespijker, 2023-10-09:
Hi @biexiangi 

Innovating on the slicing algorithm used in Cura is a complex and challenging task, but it can be feasible with the right resources, dedication, and support. Here are some key points to consider:

1. **Feasibility**: It is feasible to work on improving the slicing algorithm in Cura if you have a good understanding of 3D printing, mathematics, and programming. However, be prepared for a steep learning curve, especially if you are new to this area. To start getting a grasp of algorithms involved in transforming a 3D model you can take a look at the Wiki in this page https://github.com/Ultimaker/CuraEngine/wiki/Pipeline 

2. **Resources**: Ensure you have access to the necessary resources, such as a development environment, computing power, and 3D printers for testing your algorithm changes.

For setting up the development environment I suggest that you take a look at the Wiki page https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source which will tell you how to setup the complete development environment.

When you want to connect an external CuraEngine You might want to look in the Wiki of this repo https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source 

3. **Start Small**: Begin with small improvements or optimizations to get a feel for the codebase and the challenges involved. Gradually work your way up to more significant algorithmic changes.

4. **Testing and Validation**: Extensively test your changes to ensure they produce reliable and high-quality prints. This may involve creating a testing framework and using various 3D models to evaluate the results.

5. **Legal and Licensing**: Be aware of the licensing terms and legal considerations associated with modifying and distributing open-source software like Cura.

6. **Patience and Persistence**: Developing and innovating in this domain can be challenging, so patience and persistence are crucial. Don't be discouraged by setbacks or difficulties.

In summary, while innovating on the slicing algorithm in Cura is a challenging endeavor, it is feasible with the right approach and dedication. Keep learning, testing, and collaborating with others in the field, and you may be able to make significant contributions to the 3D printing community. Good luck with your research project!

biexiangi, 2023-10-09:
Thank you for your guidance and help. At present, I am proceeding according to the plan, completely according to the website you sent. The current problem: Steps through https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source I can't use Pycharm running, now can only be opened with Windows Powershell Cura, This problem was not solved, but it did not affect the innovation of my algorithm, and I will consider solving this problem when I modify the software interface in the future. According to the external Curaengine, https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source Currently in step by step debugging with Curaengine.exe, not very good basics, but working on it. I hope it works out for the best. Finally, thank you again for your reply.

AndyZhaoXiaohui, 2024-01-04:
> 感谢您的指导和帮助。目前我正在按照计划进行，完全按照您发的网站进行。目前的问题：通过https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source的步骤我无法使用Pycharm运行，现在只能使用Windows Powershell Cura打开，这个问题没有解决，但是并不影响我算法的创新，以后修改软件界面的时候我会考虑解决这个问题。根据外部Curaengine，https [:// github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source](https://github.com/Ultimaker/CuraEngine/wiki/Building-CuraEngine-From-Source)目前正在使用 Curaengine.exe 进行逐步调试，基础知识还不是很好，但正在努力。希望一切顺利。最后，再次感谢您的回复。
你好，我是最近开始老师让我整cura二次开发有些问题不是很懂，有些问题想要咨询您，方便知道下您的邮箱吗

AndyZhaoXiaohui, 2024-01-04:
> 
Hello, I have recently been asked by my teacher to complete the secondary development of Cura. I have some questions that I am not very familiar with, and I would like to consult with you. May I know your email

biexiangi, 2024-01-04:
> > 
> 
> Hello, I have recently been asked by my teacher to complete the secondary development of Cura. I have some questions that I am not very familiar with, and I would like to consult with you. May I know your email

Of course, we communicate and progress with each other, and my email is 1459610881@qq.com , or WeChat 1459610881，let's cheer together！