Ultimaker/CuraEngine #1990: Poor communication regarding missing max flow rate feature

L0laapk3, 2023-11-26:
Hi there, its frustrated me how little communication there has been as to why this PR for this crucial setting was suddenly struck down: #1814. All we got is this one vague reply that "<the team> has decided to take a different approach".

There have been many members asking questions and explenations, but it appears that everyone on your side has promptly unsubscribed from the thread after closing it. Several members have also created separate issues with additional questions, which appears to get the same treatment after being marked as duplicate. And surely many more have not bothered to interact after being discouraged by the current state of the threads.

To me this feature is not optional, any variable line width and height settings are unusable without it, and I would like to see this be clarified more.

Arsoth, 2023-11-26:
Don't worry, I'm sure this will be closed as a duplicate shortly as well. 

We don't know what we want after all, THEY know what we want.

anubis71x, 2023-12-06:
It's free. I'm frustrated too. But all you can do is switch to prusa slicer if you want the feature.

jellespijker, 2023-12-06:
As my colleague @casperlamboo mentioned we're investigating alternative solutions. Due to some inherit difficulties with limiting the flow jump in a naive way as proposed in the PR.

> We are afraid that, when using this new feature, the print behavior will become unpredictable. Instead we want to introduce a new feature that gives the user more insight whether flow jumps are violated, and can consequently change settings to make sure flow stays within an acceptable range.

I'm not sure how this could have been communicated better. But since we're Dutch and the Dutch are known for their directness I believe the paragraph below will be as clear and direct as possible:

Yes, we have heard you, we are considering alternative solutions, which should help you get better prints, either by warning the user of a violation of the flow or scaling the flow down proportionally. As of now the PR won't be merged, due to unpredictable behavior which it will introduce, We're looking into alternative solutions which will help guard print qualities for all scenarios.

If you really feel the need to have such a feature available, it could be done by writing an Engine plugin, this would allow people to load/use this functionality dynamically in the engine. This can be done by a community member. See the Gradual Flow plugin for inspiration: https://github.com/Ultimaker/CuraEngine_plugin_gradual_flow 

I'm locking this issue for further commenting. If people want to discus the actual Feature Request please do that in the relevant tickets. If people need help developing an Engine plugin we're also more then willing to help them out. Please raise a ticket asking for help and we will get you started.

closed 2024-01-23 by jellespijker