Ultimaker/CuraEngine #1998: Cura 5.6.0 build from the sou

TechMaverik, 2023-12-13:
Hi Greetings!

I was trying to build cura 5.6.0 from the [Source](https://github.com/Ultimaker/Cura/releases/) and I followed the steps mentioned in [Cura Build From Source](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source).

I came up with the error in building the cura engine 5.6.0 as the build fails for the same. Here with attaching the screen shots and error logs.

**Application Version**
5.6.0

**Platform**
Windows 11

**PyQt**
PyQt6

**Steps to Reproduce**
from cura branch checkout 5.6.0 and follow the steps [Source](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source)

**My Environment:**

1. OS Windows 11
2. Conan 1.60.2
3. Python 3.10.11 (also tried with 3.10.4, 3.10.6, 3.10.9)
4. CMake 3.28.0 (also tried 3.23.0)
5. MSVC 2022

**Screen shot of Error**
[image]

** Log **
[Error Log Cura Engine Failed to build 5.6.0.pdf](https://github.com/Ultimaker/CuraEngine/files/13660749/Error.Log.Cura.Engine.Failed.to.build.5.6.0.pdf)

fieldOfView, 2023-12-16:
One of the issues in the log is that there is a version conflict where different parts of the build require different versions of a dependency. `dulcificum` is new in Cura 5.6. Could you try building Cura 5.5? If that works, then at least we know where to look.

PS: WHY would you upload a log file as a paged PDF? You are not making it any easier to help you.

jellespijker, 2024-04-22:
```
ERROR: Conflict in curaengine/5.6.0:
 'curaengine/5.6.0' requires 'spdlog/1.10.0' while 'dulcificum/0.1.0-beta.1' requires 'spdlog/1.12.0'.
 To fix this conflict you need to override the package 'spdlog' in your root package.
```

Try adding `--require-override=spdlog/1.10.0@_/_` to the cmd-line to force it to resolve to a specific version, if that doesn't work for version 1.10.0 try 1.12.0
or try building from `main` or `5.7` where this dependency conflict has been fixed

closed 2024-04-22 by jellespijker