Ultimaker/CuraEngine #2004: CuraEngine - Zero print time, zero filament used, Stopping process because there are no non-empty layers.

kvostyc, 2024-01-04:
Hey guys, i have problem with CuraEngine CLI. Im using it for quite a while now and im still unable to find out what is the reason behind my problem.

Problem is that some STL files (maybe like 1 in 50) im unable to get proper gcode, it always output empty gcode with zero time printing and used filament

It look like this

**.
.
.
Stopping process because there are no non-empty layers.
.
.
.
End of gcode header.
Print time (s): 32
Print time (hr|min|s): 0h 0m 0s
Filament (mm^3): 0**

Im pretty sure that STL is not damaged because i can properly open the exact same STL file it in my Cura slicer on Windows and it slice just perfect as it is in Fusion 360 with all polygons etc, so no problem with STL fle

**Application Version**
CuraEngine 5.8

**Platform**
Linux - Terminal version of **Raspbian**

**PyQt**
5.5.12

**Display Driver**
Im using only CuraEngine in CLI

**Steps to Reproduce**
I can send you a STL file if necessary

**Actual Results**

**Expected results**
I expect proper gcode

**Additional Information**
Output gcode:
https://prnt.sc/se3nqG3eljRr

Same STL fle in CuraEngine on Windows:
https://prnt.sc/WkF3ko378FNA

kvostyc, 2024-01-09:
Is there anyone facing the same issue ?

SyllerBy, 2025-04-01:
I face it too ! :/