Ultimaker/CuraEngine #2027: CuraEngine from command line fails to accept default_value from non-leaf config json nodes

richfelker, 2024-01-30:
Running CuraEngine with the `fdmprinter.def.json` file from Cura repo is expected to work. However, it errors out with

```
Trying to retrieve setting with no value given: roofing_layer_count
```

despite the setting being present in the config json. The root cause seems to be the addition of child config nodes; without them (or with them moved out of it), fetching the value works.

pepek1910, 2024-12-10:
I encountered a similar problem with my command-line CuraEngine app. Could you clarify what you mean by 'writing non-leaf'?

fieldOfView, 2024-12-10:
By convention, CuraEngine only ever reads leaf settings (ie: settings without children). This is by design. Non-leaf settings can be overridden by more specific ("child") settings.