Ultimaker/CuraEngine #2035: CuraEngine build failed on Windows 11

Falindir, 2024-02-26:
**Application Version**
5.7.0-alpha.1

**Platform**
Windows 11

**ENV**
Cmake version 3.25.3 (3.29.0 also tried)
Conan 1.60.2
GCC G++ 12.2.0 (13.2.0 also tried)
Python 3.11.8 (Python 3.13 also tried)
VSCode 2022 with MSVC 2022

**Qt**

**PyQt**
PyQt6

**Display Driver**

**Steps to Reproduce**
Clone from main

**Actual Results**
[image]

**Expected results**

**Additional Information**

I also try on VM with Ubuntu and it's work fine but i need build from source on Windows

jellespijker, 2024-02-27:
please upload the complete output of after running the conan install command

Falindir, 2024-02-29:
PS C:\Users\jilopez\Documents\Projets\CuraEngine>** conan install . --build=missing --update**

Configuration (profile_host):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
curaengine:compiler.cppstd=20
curaengine*:compiler.cppstd=20
curaengine_plugin_infill_generate:compiler.cppstd=20
curaengine_plugin_gradual_flow:compiler.cppstd=20
curaengine_grpc_definitions:compiler.cppstd=20
scripta:compiler.cppstd=20
umspatial*:compiler.cppstd=20
dulcificum:compiler.cppstd=20
cura_settings:compiler.cppstd=20
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True
tools.build:skip_test=False

Configuration (profile_build):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=Visual Studio
compiler.cppstd=17
compiler.runtime=MD
compiler.version=17
os=Windows
os_build=Windows
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.env.virtualenv:auto_use=True
tools.gnu:define_libcxx11_abi=True
tools.build:skip_test=False

WARN: arcus/5.3.1: requirement protobuf/3.21.9 overridden by curaengine/5.7.0-alpha.1+source to protobuf/3.21.12
WARN: protobuf/3.21.12: requirement zlib/[>=1.2.11 <2] overridden by arcus/5.3.1 to zlib/1.2.12
WARN: asio-grpc/2.6.0: requirement grpc/1.54.3 overridden by curaengine/5.7.0-alpha.1+source to grpc/1.50.1
WARN: asio-grpc/2.6.0: requirement boost/1.83.0 overridden by curaengine/5.7.0-alpha.1+source to boost/1.82.0
WARN: grpc/1.50.1: requirement openssl/[>=1.1 <4] overridden by asio-grpc/2.6.0 to openssl/3.2.0
WARN: grpc/1.50.1: requirement zlib/[>=1.2.11 <2] overridden by asio-grpc/2.6.0 to zlib/1.2.12
WARN: openssl/3.2.0: requirement zlib/[>=1.2.11 <2] overridden by grpc/1.50.1 to zlib/1.2.12
WARN: asio-grpc/2.6.0: requirement grpc/1.54.3 overridden by curaengine_grpc_definitions/0.2.0-alpha.0+bb9515@ultimaker/testing to grpc/1.50.1
WARN: asio-grpc/2.6.0: requirement boost/1.83.0 overridden by curaengine_grpc_definitions/0.2.0-alpha.0+bb9515@ultimaker/testing to boost/1.82.0
WARN: grpc/1.50.1: requirement openssl/[>=1.1 <4] overridden by asio-grpc/2.6.0 to openssl/3.2.0
WARN: grpc/1.50.1: requirement zlib/[>=1.2.11 <2] overridden by asio-grpc/2.6.0 to zlib/1.2.12
WARN: spdlog/1.12.0: requirement fmt/10.2.1 overridden by curaengine/5.7.0-alpha.1+source to fmt/10.1.1
clipper/6.4.2@ultimaker/stable: WARN: Can't update, no package in remote
zlib/1.2.12: WARN: Can't update, no package in remote
openssl/3.2.0: WARN: Can't update, no package in remote
protobuf/3.21.12: WARN: Can't update, no package in remote
arcus/5.3.1: WARN: Can't update, no package in remote
grpc/1.50.1: WARN: Can't update, no package in remote
curaengine_grpc_definitions/0.2.0-alpha.0+bb9515@ultimaker/testing: WARN: Can't update, no package in remote
[... 471 lines cut ...]
FAILED: CMakeFiles/_CuraEngine.dir/src/InterlockingGenerator.cpp.obj
C:\mingw64\bin\c++.exe -DAGRPC_BOOST_ASIO -DARCUS -DBUILD_TESTS -DCARES_STATICLIB -DCURA_ENGINE_VERSION=\"5.7.0-alpha.1+source\" -DENABLE_PLUGINS -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSTB_TEXTEDIT_KEYTYPE=unsigned -IC:/Users/jilopez/Documents/Projets/CuraEngine/include -IC:/Users/jilopez/Documents/Projets/CuraEngine/build/Release -isystem C:/.conan/87e09e/1/include -isystem C:/.conan/acaa8c/1/include -isystem C:/.conan/1433558/1/include -isystem C:/.conan/aa8dc886/1/include -isystem C:/.conan/cf8014e0/1/include -isystem C:/.conan/f288abc3/1/include -isystem C:/.conan/b3f7f481/1/include -isystem C:/.conan/03375768/1/include -isystem C:/.conan/512969eb/1/include -isystem C:/.conan/7e46b946/1/include -isystem C:/.conan/502e6455/1/include -isystem C:/.conan/fc7cde3f/1/include -isystem C:/.conan/5069943e/1/include -isystem C:/.conan/b9181f2d/1/include -isystem C:/.conan/cbb29a46/1/include -isystem C:/.conan/4a802f/1/include -isystem C:/.conan/9521b8cf/1/include -isystem C:/.conan/1f3189/1/include -isystem C:/.conan/abcf006c/1/include -isystem C:/.conan/1669c878/1/include -isystem C:/.conan/f67448e7/1/include -isystem C:/.conan/73221bc/1/include -isystem C:/.conan/687b4214/1/include -O3 -DNDEBUG -std=c++20 -pthread /permissive- -MD -MT CMakeFiles/_CuraEngine.dir/src/InterlockingGenerator.cpp.obj -MF CMakeFiles\_CuraEngine.dir\src\InterlockingGenerator.cpp.obj.d -o CMakeFiles/_CuraEngine.dir/src/InterlockingGenerator.cpp.obj -c C:/Users/jilopez/Documents/Projets/CuraEngine/src/InterlockingGenerator.cpp
c++.exe: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
[20/190] Building CXX object CMakeFiles/_CuraEngine.dir/src/infill.cpp.obj
FAILED: CMakeFiles/_CuraEngine.dir/src/infill.cpp.obj
C:\mingw64\bin\c++.exe -DAGRPC_BOOST_ASIO -DARCUS -DBUILD_TESTS -DCARES_STATICLIB -DCURA_ENGINE_VERSION=\"5.7.0-alpha.1+source\" -DENABLE_PLUGINS -DNOMINMAX -DSPDLOG_COMPILED_LIB -DSPDLOG_FMT_EXTERNAL -DSTB_TEXTEDIT_KEYTYPE=unsigned -IC:/Users/jilopez/Documents/Projets/CuraEngine/include -IC:/Users/jilopez/Documents/Projets/CuraEngine/build/Release -isystem C:/.conan/87e09e/1/include -isystem C:/.conan/acaa8c/1/include -isystem C:/.conan/1433558/1/include -isystem C:/.conan/aa8dc886/1/include -isystem C:/.conan/cf8014e0/1/include -isystem C:/.conan/f288abc3/1/include -isystem C:/.conan/b3f7f481/1/include -isystem C:/.conan/03375768/1/include -isystem C:/.conan/512969eb/1/i [... cut ...] /1/include -isystem C:/.conan/687b4214/1/include -O3 -DNDEBUG -std=c++20 -pthread /permissive- -MD -MT CMakeFiles/_CuraEngine.dir/src/GCodePathConfig.cpp.obj -MF CMakeFiles\_CuraEngine.dir\src\GCodePathConfig.cpp.obj.d -o CMakeFiles/_CuraEngine.dir/src/GCodePathConfig.cpp.obj -c C:/Users/jilopez/Documents/Projets/CuraEngine/src/GCodePathConfig.cpp
c++.exe: fatal error: cannot specify '-o' with '-c', '-S' or '-E' with multiple files
compilation terminated.
ninja: build stopped: subcommand failed.

Falindir, 2024-02-29:
The error is truly mysterious. I've tested rebuild from zero also with different version of gcc and cmake and python. Even with CuraEngine 5.5 and 5.6 the same error. It comes from my system but I can't see where.

Falindir, 2024-03-11:
Up maybe.

Falindir, 2024-03-12:
Well, I don't understand why, but I solved my problem with `conan create ...` directly.

closed 2024-03-12 by Falindir

jellespijker, 2024-03-14:
it seems you're building with mingw.

```
C:\mingw64\bin\c++.exe
```

maybe you have the variables CC and CXX defined, if so unset these

DDAndLL9698, 2025-01-22:
@jellespijker Can the official release a markdown file to teach new user to build cura and curaengine step by step? i follow the github wiki  but wrong.  i had to google it to find the way. but it is not easy,  there is too little relevant information.  if we have an official  markdown  file ,  no need to ask your guys on github. what do you think?

jellespijker, 2025-01-22:
@wawanbreton or @rburema can either of you take a look at this request

rburema, 2025-01-22:
At the moment, the leading source should be: https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source -- this is the updated workflow from when we switched to Conan 2.

Then, if you want to work on the engine (or some other component that isn't 'front-end' Cura, like Uranium), you put the package in Conan's 'editable' mode: https://docs.conan.io/2/reference/commands/editable.html

On Windows, we don't tend to use GCC, but MSVC instead. If there are conflicting versions of C++ compilers in the PATH that might cause problems.