Ultimaker/CuraEngine #2070: Unable to compile on Ubuntu 22.04

grepwood, 2024-04-22:
I have Conan 1.60.2, GCC 12, CMake 3.25.1, Python 3.10, and I run this:
```
conan install . --build=missing --update
cmake --preset release
```
Except for having to hack my way around https://github.com/Ultimaker/CuraEngine/issues/2068, everything is hunky dory up to this point. But when I run
```
cmake --build --preset release
```
All hell breaks loose as evidenced in the attached log.
[curaengine-5.7.0-build.log](https://github.com/Ultimaker/CuraEngine/files/15066874/curaengine-5.7.0-build.log)

jellespijker, 2024-04-22:
try updating gnu-gcc to version 13 or higher
```
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt install g++-13 gcc-13 -y
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-13 13
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-13 13
```
run conan profile detection again such that it uses the gcc-13

```
conan profile new default --detect --force
```

I will update the conanfiles such that the will fail with gcc-12 or lower

closed 2024-04-22 by jellespijker

grepwood, 2024-04-22:
I figured out how to build it with GCC 12:
```
CONAN_CONFIG_VERSION='d79fe6fe1878f09b583e04b873544cec65cbac99' && conan config install https://github.com/Ultimaker/conan-config/archive/${CONAN_CONFIG_VERSION}.zip
conan profile new default --detect --force
conan profile update settings.compiler.libcxx=libstdc++11 default
conan profile update settings.compiler.cppstd=gnu20 default
```
The last line is most important.