Ultimaker/CuraEngine #2071: Please start building each version before merging it to the main branch

grepwood, 2024-04-22:
We've barely had 5.7.0 for a month and from what I see 5.7.1 is about to be released according to https://github.com/Ultimaker/CuraEngine/actions/runs/8782671105 but what's this? The pipeline doesn't compile anything. So this probably explains why this project has so many build issues - the issues do not become apparent to the developers before merging into the master branch.
The issues, just to name a few are:
- 5.0.0 https://github.com/Ultimaker/CuraEngine/issues/2017
- 5.6.0 https://github.com/Ultimaker/CuraEngine/issues/1998
- 5.7.0 https://github.com/Ultimaker/CuraEngine/issues/2070
- 5.7.0-alpha1 https://github.com/Ultimaker/CuraEngine/issues/2035
- 5.7.0-alpha1 https://github.com/Ultimaker/CuraEngine/issues/2019
- 5.5 and 5.6 https://github.com/Ultimaker/CuraEngine/issues/1991

Why is it like this?:
- MacOS build skipped https://github.com/Ultimaker/CuraEngine/actions/runs/8782671105/job/24097251782
- Windows build skipped https://github.com/Ultimaker/CuraEngine/actions/runs/8782671105/job/24097251562
- Linux build skipped https://github.com/Ultimaker/CuraEngine/actions/runs/8782671105/job/24097251340

If the GitHub project would be configured to require that each PR must produce binaries, then the quality of the project would take a dramatic increase.

jellespijker, 2024-04-22:
pr's to main and release branches have unit test, which means that the binaries for these are build, from that branch before they are merged.

furthermore due to CuraEngine being a deps of Cura, there are nightlies  being build for our QA each day, both from main as the latest release  branch.

Additional each ticket implemented by us has a corresponding build from the working  branch made by our QA before merging to main or release. 

Additional statistical analysis tools are also run on each open PR for gcode output, speed benchmarks, stress tests. in otherwords our CI/CT is rather comprehensive. 

the issue you're referring to were simply the results of GitHub runner images being updated and missing certain system dependencies.

These have been adressed.
closing this ticket

closed 2024-04-22 by jellespijker

grepwood, 2024-04-22:
Thank you, this clarifies. For anyone looking for those builds, where can they be found?

jellespijker, 2024-04-22:
If you mean the stress tests, benchmarks and regression test these can be found here:
- https://ultimaker.github.io/CuraEngineBenchmarks/dev/bench/
- https://ultimaker.github.io/CuraEngineBenchmarks/dev/gcodeanalyzer/
- https://ultimaker.github.io/CuraEngineBenchmarks/dev/stress_bench/

The unit tests themself are tied to the PR's adn require a :heavy_check_mark: before you can merge.

If you mean the actual binaries, these are upload to our jfrog artifactory for consumption by downstream projects (e.q. Cura) and are automaticaly used when doing a conan install for that project itself. If you tried that but you see that CuraEngine is still being build, that is because of ABI compatabilities, you're likely using an other architecture, os, compiler then we use in our build system our on our development systems