Ultimaker/CuraEngine #2185: 2 models, 2 tools one at the time

pepek1910, 2024-12-10:
I’m using CuraEngine from the command line and need to generate G-code for two models, each printed with a different tool, one at a time. I understand this isn’t directly possible in the Cura Ultimaker software. Is there a way to bypass this limitation using CuraEngine from the command line (some combination of -g -m --next)?
Additionally, does anyone know if this feature might be included in future versions of Cura, such as 6.0?

fieldOfView, 2024-12-10:
> I understand this isn’t directly possible in the Cura Ultimaker software

On the contrary, this is possible in Cura, and has been since Cura 2.3 as far as I know.

Couldn't be easier:
* load a model and duplicate it (or load two models)
* set print sequence to "One at a time"
* select the second model
* press the second extruder icon in the toolbar on the left

There is a general limitation of "One at a time" printing that the available print height is reduced, to prevent collissions between the print gantry and a previous print.

pepek1910, 2024-12-12:
Thank you! That’s exactly what I was looking for. I followed your instructions, and the 'Print Sequence' parameter is now visible.

However, I’ve encountered another issue. When I assign the first tool to the first model and the second tool to the second model, the slicing result includes only the full first model, while the second model is reduced to just the brim. Additionally, if the second model is taller than the first, it starts printing brim and then only part above the height of the first model, so the bottom part is missing.

I understand that I changed the settings and  it might cause issues, but do you know why this happens? Which parameter should I adjust to ensure both models are fully printed with their assigned tools