Ultimaker/CuraEngine #2194: `find_package(standardprojectsettings REQUIRED)` Fails in `CMakeLists.txt`

FreeFallingSnow, 2024-12-31:
### **Description:**

I am trying to build the CuraEngine project using Visual Studio 2022 and CMake. During the configuration step, I encounter an error related to the `find_package(standardprojectsettings REQUIRED)` command in `CMakeLists.txt`. The error indicates that the `standardprojectsettings` package cannot be found.

------

### **Steps to Reproduce:**

1. Set up the Conan environment:

   ```powershell
   python -m venv cura_venv
   cura_venv\Scripts\activate.ps1
   
   pip install conan==2.7.0
   pip install CMake
   pip install Ninja
   pip install sip==6.5.1
   
   conan config install https://github.com/ultimaker/conan-config.git
   conan profile detect --force
   
   git clone https://github.com/Ultimaker/CuraEngine.git
   cd CuraEngine
   
   conan install . --build=missing --update
   conan install . --build=missing -s build_type=Debug -o curaengine/*:enable_testing=True
   ```

2. Open the `CuraEngine` folder in Visual Studio 2022.

   Error are found:

   ```
   在路径“d:\graduationdesignprogram\cura\curaengine\cmakepresets.json”中未找到 Include		D:\GraduationDesignProgram\Cura\CuraEngine\CMakeUserPresets.json	1	
   ```

   ```
   Translation:
   Include not found in the path "d:\graduationdesignprogram\cura\curaengine\cmakepresets.json" D:\GraduationDesignProgram\Cura\CuraEngine\CMakeUserPresets.json 1
   ```

   Solve the problem in this way:

   ```
   Copy-Item CMakeUserPresets.json CMakePresets.json
   ```

   

------

### **Expected Behavior:**

The configuration should successfully locate and include the `standardprojectsettings` package.

------

### **Actual Results**

The following error is generated:

```
CMake Error at D:\GraduationDesignProgram\Cura\CuraEngine\CMakeLists.txt:7 (find_package):
  By not providing "Findstandardprojectsettings.cmake" in CMAKE_MODULE_PATH
  this project has asked CMake to find a package configuration file provided
  by "standardprojectsettings", but CMake did not find one.

  Could not find a package configuration file provided by
  "standardprojectsettings" with any of the following names:

    standardprojectsettingsConfig.cmake
    standardprojectsettings-config.cmake

  Add the installation prefix of "standardprojectsettings" to
  CMAKE_PREFIX_PATH or set "standardprojectsettings_DIR" to a directory
  containing one of the above files.  If "standardprojectsettings" provides a
  separate development package or SDK, be sure it has been installed.	CuraEngine	D:\GraduationDesignProgram\Cura\CuraEngine\CMakeLists.txt	7		

```

------

### **Environment:**

- **OS:** Windows 11 (64-bit)
- **Visual Studio Version:** 2022

------

### **Troubleshooting Steps Taken:**

1. Verified that `conan config install` and `conan install` were completed successfully.
2. Checked for the presence of `standardprojectsettings` in the Conan cache.

------

### **Request:**

Could you provide guidance on resolving this issue? Is there an update or specific setup step required to correctly include the `standardprojectsettings` package in the build configuration?

------

### **Additional Information:**

- Let me know if additional logs or files are needed to assist with troubleshooting.

beyond-space21, 2025-01-04:
hello. I stuck with the same error, got any solution yet?

FreeFallingSnow, 2025-01-04:
It hasn't been solved yet, if you have a solution later, please reply to me, thank you
@beyond-space21

B-S-P, 2025-01-04:
I've ran into the same issue, I've been able to work around it to successfully build using "x64 Native Tools Command Prompt for VS 2022"; but not yet been able to get it to work in visual studio.

'x64 Native Tools' are typically found under `C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Visual Studio 2022\Visual Studio Tools\VC`.

Internally it seems to just seems to be a shortcut to run `%comspec% /k "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat"`, where `%comspec%`  is a built in environment variable that resolves to your command shell, likely `C:\WINDOWS\system32\cmd.exe`.
So you should be able to run `C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat` directly from a CMD prompt.

In the x64 Tools prompt you should be able to run the commands from your steps to reproduce, activating .bat instead of ps1. Followed by invoking cmake, in full that'll look like this:
```
REM Create venv for cura dependencies then activate
python -m venv cura_venv
cura_venv\Scripts\activate.bat

REM install conan
pip install conan

REM Setup conan-config
conan config install https://github.com/ultimaker/conan-config.git

REM detect profile
conan profile detect

conan install . --build=missing --update
conan install . --build=missing -s build_type=Debug -o 'curaengine/*:enable_testing=True'

REM Run these for release build
cmake --preset conan-release
cmake --build --preset conan-release

REM run these for debug build
cmake --preset conan-build
cmake --build --preset conan-build
```

I was led to try the above by the comments in the following related issue, especially the comment I've linked directly:
https://github.com/Ultimaker/CuraEngine/issues/1770#issuecomment-1350582607

FreeFallingSnow, 2025-01-05:
I tried your approach, the difference is that I used Developer PowerShell for VS 2022 and of course cura_venv\Scripts\activate.ps1 to enter the virtual environment. I successfully built the release version, but an error occurred while running cmake --build --preset conan-debug. Do you know the possible cause?

[log.txt](https://github.com/user-attachments/files/18309009/log.txt)

Of course, I used the x64 Native Tools Command Prompt for VS 2022 to generate errors
   
[log in cmd.txt](https://github.com/user-attachments/files/18309010/log.in.cmd.txt)

> I've ran into the same issue, I've been able to work around it to successfully build using "x64 Native Tools Command Prompt for VS 2022"; but not yet been able to get it to work in visual studio.我遇到了同样的问题，我已经能够解决它以使用“VS 64 的 x2022 原生工具命令提示符”成功构建;但还无法让它在 Visual Studio 中工作。
> 
> 'x64 Native Tools' are typically found under `C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Visual Studio 2022\Visual Studio Tools\VC`.“x64 本机工具”通常位于 `C:\ProgramData\Microsoft\Windows\Start Menu\Programs\Visual Studio 2022\Visual Studio Tools\VC` 下。
> 
> Internally it seems to just seems to be a shortcut to run `%comspec% /k "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat"`, where `%comspec%` is a built in environment variable that resolves to your command shell, likely `C:\WINDOWS\system32\cmd.exe`.在内部，它似乎只是一个 run `%comspec% /k "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat"` 的快捷方式，其中 `%comspec%` 是一个内置环境变量，可以解析为您的命令 shell，可能是 `C：\WINDOWS\system32\cmd.exe`。 So you should be able to run `C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat` directly from a CMD prompt.因此，您应该能够直接从 CMD 提示符运行 `C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars64.bat` 。
> 
> In the x64 Tools prompt you should be able to run the commands from your steps to reproduce, activating .bat instead of ps1. Followed by invoking cmake, in full that'll look like this:在 x64 Tools 提示符中，您应该能够运行步骤中的命令进行重现，激活 .bat 而不是 ps1。然后调用 cmake，完整如下所示：
> 
> ```
> REM Create venv for cura dependencies then activate
> python -m venv cura_venv
> cura_venv\Scripts\activate.bat
> 
> REM install conan
> pip install conan
> 
> REM Setup conan-config
> conan config install https://github.com/ultimaker/conan-config.git
> 
> REM detect profile
> conan profile detect
> 
> conan install . --build=missing --update
> conan install . --build=missing -s build_type=Debug -c tools.build:skip_test=False
> 
> REM Run these for release build
> cmake --preset conan-release
> cmake --build --preset conan-release
> 
> REM run these for debug build
> cmake --preset conan-debug
> cmake --build --preset conan-debug
> ```
> 
> NOTE: To build with tests in the latest versions you must use `-c tools.build:skip_test=False` in place of `-o 'curaengine/*:enable_testing=True'`注意：要使用最新版本中的测试进行构建，您必须使用 `-c tools.build:skip_test=False` 代替 `-o 'curaengine/*:enable_testing=True'`
> 
> I was led to try using the approach above by the comments in the following related issue, especially the comment I've linked directly:以下相关问题中的评论引导我尝试使用上述方法，尤其是我直接链接的评论： [#1770 (comment)](https://github.com/Ultimaker/CuraEngine/issues/1770#issuecomment-1350582607)

DDAndLL9698, 2025-01-22:
这个需要设置一下CMAKE_PREFIX_PATH

FreeFallingSnow, 2025-02-11:
能详细说一下吗，谢谢

> [@FreeFallingSnow](https://github.com/FreeFallingSnow) 这个需要设置一下CMAKE_PREFIX_PATH