Ultimaker/CuraEngine #2195: Unable to build on ArchLinux / Failed requirements

corbin-c, 2025-01-03:
Hi there,

I'm trying to build CuraEngine from source on my machine.

The build process with conan seems kinda broken.

As per the wiki, I've installed conan in version 1.65.0. I'm running this in a venv so it doesn't mess with my whole setup:

```bash
$ python -m venv cura_venv
$ source cura_venv/bin/activate
$ pip install conan==1.65
```

Now I can assert conan is running fine:

```bash
$ conan --version
Conan version 1.65.0
```

The profile setup runs fine. But then, when trying to install, I get the following:

```bash
$ conan install . --build=missing --update
WARN: **************************************************
WARN: *** Conan 1 is legacy and on a deprecation path **
WARN: *********** Please upgrade to Conan 2 ************
WARN: **************************************************
ERROR: Error reading '/home/user/.conan/profiles/cura.jinja' profile: ConfigParser: Unrecognized field 'replace_requires'
```

As far as my understanding of conan goes, `replace_requires` is a conan >2 option. This seems in contradiction with the specified version (1.65.0). This is weird, but I found a workaround by removing the section altogether in the cura profile. 

Anyway, let's try and re-build!

```bash
$conan install . --build=missing --update

WARN: **************************************************
WARN: *** Conan 1 is legacy and on a deprecation path **
WARN: *********** Please upgrade to Conan 2 ************
WARN: **************************************************
Configuration (profile_host):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.cppstd=17
compiler.libcxx=libstdc++11
compiler.version=14
os=Linux
os_build=Linux
curaengine*:compiler.cppstd=20
curaengine_plugin_infill_generate*:compiler.cppstd=20
curaengine_plugin_gradual_flow*:compiler.cppstd=20
curaengine_grpc_definitions*:compiler.cppstd=20
scripta*:compiler.cppstd=20
umspatial*:compiler.cppstd=20
dulcificum*:compiler.cppstd=20
curator/*:compiler.cppstd=20
[options]
asio-grpc:local_allocator=recycling_allocator
boost:header_only=True
clipper:shared=True
cpython:shared=True
cpython:with_curses=False
cpython:with_tkinter=False
dulcificum:shared=True
grpc:csharp_plugin=False
grpc:node_plugin=False
grpc:objective_c_plugin=False
grpc:php_plugin=False
grpc:python_plugin=False
grpc:ruby_plugin=False
openssl:shared=True
pyarcus:shared=True
pynest2d:shared=True
pysavitar:shared=True
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.gnu:define_libcxx11_abi=True
tools.build:skip_test=True

Configuration (profile_build):
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.cppstd=17
compiler.libcxx=libstdc++11
compiler.version=14
os=Linux
os_build=Linux
curator/*:compiler.cppstd=20
[options]
[build_requires]
[env]
[conf]
tools.cmake.cmaketoolchain:generator=Ninja
tools.gnu:define_libcxx11_abi=True
tools.build:skip_test=True

scripta/0.1.0@ultimaker/testing: Not found in local cache, looking in remotes...
scripta/0.1.0@ultimaker/testing: Trying with 'conancenter'...
ERROR: Failed requirement 'scripta/0.1.0@ultimaker/testing' from 'conanfile.py (curaengine/5.10.0-alpha.0+source)'
ERROR: Unable to find 'scripta/0.1.0@ultimaker/testing' in remotes
```

As you can see, the build process stops because of the missing dependency `scripta/0.1.0@ultimaker/testing`.

I have tried a few things, in particular removing the cura remote with `conan remote remove cura` or manually adding a remote: `conan remote add cura-conan https://cura.jfrog.io/artifactory/api/conan/cura-conan`, all without success.

Any help is very welcome!

corbin-c, 2025-01-04:
For anyone facing the same issue, I found a workaround. One has to use the old conan config (pre v.2). So, instead of doing this step from the wiki:

```
conan config install https://github.com/ultimaker/conan-config.git
```

Do the following:
```bash
git clone https://github.com/ultimaker/conan-config.git
cd conan-config
git checkout 3226488623c642b40ca7ce3f62d3f33de046d11e
cd ..
conan config install ./conan-config
```

Once this is done, I had to alter the remotes too, by adding conan center and removing cura private, so that `~/.conan/remotes.json` looks like this:

```json
{
 "remotes": [
  {
   "name": "conancenter",
   "url": "https://center.conan.io",
   "verify_ssl": true
  },
  {
   "name": "cura",
   "url": "https://cura.jfrog.io/artifactory/api/conan/cura-conan-dev",
   "verify_ssl": true
  }
 ]
}
```

I hope this helps someone, somewhere!

Qu-Dasheng, 2025-01-06:
I have solved as well, thank you.

DiegoHerrera1890, 2025-01-21:
Hi, thank you for posting.

I encountered this error while trying to install CuraEngine 5.9.0, but I couldn't resolve it.
I built the latest CuraEngine:

```
Cura_SteamEngine version 5.10.0-alpha.0

Copyright (C) 2024 Ultimaker

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
GNU Affero General Public License for more details.
```
However, when running the following command:

`./build/Release/CuraEngine slice -v -p -j /home/diego/Downloads/Cura-5.8.0/resources/definitions/fdmprinter.def.json -j /home/diego/Downloads/Cura-5.8.0/resources/definitions/fdmextruder.def.json -l /home/diego/Downloads/curaengine/3dbenchy.stl -o /home/diego/Downloads/curaengine/output/output.gcode`

I received this error:

```
[2025-01-21 19:12:41.767] [error] Trying to retrieve setting with no value given: roofing_layer_count
[2025-01-21 19:12:41.767] [debug] Processing skins and infill layer 64 of 478
[2025-01-21 19:12:41.768] [debug] Processing skins and infill layer 13 of 478
[2025-01-21 19:12:41.768] [error] Trying to retrieve setting with no value given: roofing_layer_count
[2025-01-21 19:12:41.768] [debug] Processing skins and infill layer 33 of 478
Segmentation fault (core dumped)
```
I suspect this error occurs because the `fdmprinter.def.json` file is from Ultimaker Cura 5.9.0, while the CuraEngine version I built is the latest one, 5.10.0, and not 5.9.0.

Do you have any advice on how to address this issue?

corbin-c, 2025-01-21:
Hi,

```
I suspect this error occurs because the fdmprinter.def.json file is from Ultimaker Cura 5.9.0, while the CuraEngine version I built is the latest one, 5.10.0, and not 5.9.0.
```

Indeed the problem is because you're using definition files from other Cura version. You need to reference the definitions from the same CuraEngine version you're running.

Try downloading them directly from the Cura repository: https://github.com/Ultimaker/Cura/tree/main/resources/definitions

Qu-Dasheng, 2025-01-21:
beacuse the bad network,I download the conan config from china platform, i found that it's not the lastest version, but the CuraEngine Source is lastest. When i get the conan config from github directly, no probelm occured.