Ultimaker/CuraEngine #2216: CuraEngine does not generate results as GUI

DiegoHerrera1890, 2025-03-19:
Dear Conmunity.
I am experiencing significant differences in how CuraEngine CLI and Cura GUI generate inner walls, and outer walls even when using the exact same parameter settings.

Steps to Reproduce:
I use the following parameters in CuraEngine CLI:
```
acceleration_enabled=False
adaptive_layer_height_variation=0.3
adaptive_layer_height_variation_step=0.05
adhesion_type=brim
alternate_extra_perimeter=True
bottom_layers=7
bottom_skin_expand_distance=6.0
bottom_skin_preshrink=6.0
bottom_thickness=1.0
brim_gap=0.0
brim_inside_margin=4.8
brim_location=outside
brim_smart_ordering=True
brim_width=8.0
center_object=True
clean_between_layers=False
cool_during_extruder_switch=unchanged
cool_fan_enabled=True
cool_fan_full_at_height=0.4
cool_fan_full_layer=2
cool_fan_speed=0.0
cool_fan_speed_0=0.0
cool_fan_speed_max=0.0
cool_fan_speed_min=0.0
cool_lift_head=False
cool_min_layer_time=7.0
cool_min_layer_time_fan_speed_max=10.0
cool_min_speed=20.0
cool_min_temperature=240.0
definition=fdmprinter
expand_skins_expand_distance=6.0
fill_outline_gaps=False
gradual_infill_steps=0.0
gradual_support_infill_steps=0.0
hole_xy_offset=0.0
infill_before_walls=False
infill_material_flow=100.0
infill_multiplier=1
infill_overlap=10.0
infill_overlap_mm=0.5
infill_pattern=gyroid
infill_randomize_start_location=False
infill_sparse_density=14.0
infill_sparse_thickness=0.4
infill_support_enabled=False
infill_wipe_dist=5.0
initial_bottom_layers=33
inset_direction=inside_out
intent_category=default
ironing_enabled=False
jerk_enabled=False
layer_height=0.4
layer_height_0=0.4
layer_start_x=0.0
layer_start_y=0.0
machine_depth=300
machine_height=600
machine_nozzle_size=1
machine_width=300
material_bed_temperature=70.0
material_final_print_temperature=280.0
material_flow=100.0
material_flow_layer_0=100.0
material_initial_print_temperature=280.0
material_print_temperature=280.0
material_print_temperature_layer_0=280.0
material_shrinkage_percentage=100.0
material_shrinkage_percentage_xy=100.0
material_shrinkage_percentage_z=100.0
max_skin_angle_for_expansion=90.0
meshfix_maximum_deviation=0.05
meshfix_maximum_travel_resolution=0.5
min_bead_width=1.0
min_infill_area=0.0
min_skin_width_for_expansion=0.0
minimum_support_area=0.0
name=POPRINT_PETG_40_Z_HOP
optimize_wall_printing_order=True
position=0
prime_tower_flow=100.0
quality_type=verydraft
retract_at_layer_change=True
retraction_amount=0.5
retraction_combing=off
retraction_count_max=2
retraction_enable=True
retraction_extra_prime_amount=0.25
retraction_extrusion_window=5
retraction_hop=1.0
retraction_hop_enabled=True
retraction_hop_only_when_collides=False
retraction_min_travel=line_width*2
retraction_speed=20.0
retraction_prime_speed=retraction_speed
retraction_retract_speed=20.0
roofing_layer_count=0
setting_version=23
skin_angles=[15,30,45,60,90,105,120,135,150,165,180]
skin_edge_support_layers=0.0
skin_edge_support_thickness=0.0
skin_material_flow=100.0
skin_material_flow_layer_0=100.0
skin_monotonic=False
skin_no_small_gaps_heuristic=False
skin_outline_count=0
skin_overlap=5.0
skin_overlap_mm=0.4
skin_preshrink=6.0
skirt_brim_material_flow=100.0
skirt_brim_minimal_length=250.0
skirt_brim_speed=60.0
small_skin_on_surface=False
small_skin_width=2.4
speed_equalize_flow_width_factor=100
speed_infill=80.0
speed_layer_0=60.0
speed_print=80.0
speed_print_layer_0=60.0
speed_slowdown_layers=1
speed_support=140.0
speed_support_infill=140
speed_topbottom=80.0
speed_travel=300.0
speed_travel_layer_0=100
speed_wall=60.0
speed_wall_0=80.0
speed_wall_0_roofing=80.0
speed_wall_x=80.0
speed_wall_x_roofing=80.0
speed_z_hop=25.0
support_angle=40.0
support_bottom_stair_step_height=0.1
support_bottom_stair_step_width=5.0
support_brim_enable=True
support_connect_zigzags=True
support_enable=True
support_infill_angles=[40]
support_infill_rate=30.0
support_infill_sparse_thickness=0.4
support_join_distance=2.0
support_material_flow=100.0
support_offset=1.6
support_pattern=zigzag
support_structure=normal
support_top_distance=0.1
support_type=buildplate
support_use_towers=False
support_wall_count=0.0
support_xy_distance=0.1
support_xy_distance_overhang=0.1
support_xy_overrides_z=z_overrides_xy
support_z_distance=0.1
support_z_seam_away_from_model=True
support_z_seam_min_distance=2.4
top_bottom_pattern=zigzag
top_bottom_pattern_0=zigzag
top_bottom_thickness=1.0
top_layers=7
top_skin_expand_distance=6.0
top_skin_preshrink=6.0
top_thickness=1.0
travel_avoid_distance=2
travel_avoid_other_parts=False
travel_avoid_supports=False
travel_retract_before_outer_wall=True
type=quality_changes
wall_0_inset=0.2
wall_0_material_flow=100.0
wall_0_material_flow_layer_0=100.0
wall_0_material_flow_roofing=100.0
wall_0_wipe_dist=machine_nozzle_size/2
wall_distribution_count=1000
wall_material_flow=100.0

wall_transition_angle=30.0
wall_transition_filter_deviation=0.0
wall_x_material_flow=100.0
wall_x_material_flow_layer_0=100.0
wall_x_material_flow_roofing=100.0
xy_offset=0.0
xy_offset_layer_0=0.0
z_seam_corner=z_seam_corner_weighted
z_seam_on_vertex=False
z_seam_position=left
z_seam_relative=True
z_seam_type=back
z_seam_x=-150.0
z_seam_y=0.0
zig_zaggify_infill=True
material_diameter=1.75
min_wall_line_width=1.2
line_width = 1.2
infill_line_width=1.2
wall_thickness=10
wall_line_width=1.3
wall_line_count=5
wall_line_width_0=1.2
wall_line_width_x=1.2
wall_transition_length=0.1
support_line_width=1.2
infill_line_distance=8.5714
brim_line_count=20
support_line_distance=4.0
support_initial_layer_line_distance=4.0
initial_layer_line_width_factor=100.0
skirt_brim_line_width=1
skin_line_width=1.2
```

I generate the G-code using CuraEngine CLI.
I compare the result with the G-code generated in Cura GUI using the exact same parameters.
**Expected Results:**
The inner walls (green lines) should be generated identically in CLI and GUI.
There should be no gaps between inner walls in either version.
**Actual Results:**
In CuraEngine CLI, inner walls have visible gaps between them.
In Cura GUI, inner walls are correctly generated with no gaps.
NOTE:

In Cura GUI, I checked that the values match those listed above, but CuraEngine CLI does not produce the same output or even a similar one. I used the last version of both CuraEngine and Ultimaker Cura.

In the attached image, left side is generated by CuraEngine, Right side is generated by Ultimaker Cura. Using teh same parameters.
[image]

Diego1890, 2025-10-29:
Thank you for your comment. It can be closed for now since I am not working on that for now but I might open it again in the future please.

fieldOfView, 2025-10-30:
While there are obvious differences in both slice results (line width and bed adhesion seem dramatically different), one thing to consider is that the regular preview in Cura can look different from the preview of a gcode file. There is some information not present in gcode files that is present right after slicing. The preview right after slicing is actually not a preview of the gcode but based on a separate stream of (more efficient) data.